VIP-LES/EosPayload #22: Reduce Unnecessary Threading

DarylDohner, 2022-11-13:
To reduce resource strain on the beaglebones, add config methods like command_thread_enabled() and read_thread_enabled() so we aren't spawning unnecessary threads that just spin forever.  Ensure health check accounts for this

thomasmholder, 2022-11-15:
If this is an issue, a bandaid fix could be to slow the spin function down even further.

DarylDohner, 2022-11-16:
Current ideas:
1. Define a method for non-blocking things that can execute in the main thread.  Ie, registering a callback
2. a choice:  either config funcs for read_thread_enabled / command_thread_enabled, or we completely change how we do threading and just expose a register_thread method that can be invoked during setup (instead of providing read/cmd thread out-of-the-box)

DarylDohner, 2022-11-16:
Also this ticket should include disabling the test drivers

closed 2022-12-07 by DarylDohner