VIP-LES/EosPayload #46: Select Sensor Suit

David-Rey, 2023-02-03:
Select Sensors for telemetry.
Google Sheet: https://docs.google.com/spreadsheets/d/1kyEElZ4ADUigGbQcuU-MNJgED3wliDuhFE-724LtRRI/edit#gid=0

A few sentences of why you picked each sensor

cjkerr128, 2023-02-07:
GPS: Adafruit Ultimate GPS
https://www.adafruit.com/product/5440
Weight: 8.5g
GPS & GLONASS SUPPORT
5V logic safe
Current: 30mA current draw 
Power Usage:  ~29 mA 

IMU: BNO055
https://www.adafruit.com/product/4646
combines MEMS accelerometer, magnetometer, and gyroscope to deliver actual 3D Space Orientation
Outputs: Absolute Orientation (by Euler vector or quaternion) Angular Velocity Vectors, Acceleration vector, Magnetic Field Strength Vector, Gravity Vector
I2C Support and is 3/5V logic safe, interface with Arduino or Beaglebone

Pressure: MS8607
https://www.adafruit.com/product/4716
I2C Support is 3/5V logic safe, interface with Arduino or Beaglebone
Operating Range: 10 to 2000 mbar
Pressure accuracy is +- 2hPa, low power consumption with usage as low as 0.78 µA

DarylDohner, 2023-02-07:
Could you put this in spreadsheet form on the drive in the software folder?

DarylDohner, 2023-02-10:
approved

DarylDohner, 2023-02-10:
https://docs.google.com/spreadsheets/d/1iYUvNZitJe_nFyax71T-RW-lnOEBMcUu8k-r4sQGpAE/edit#gid=0

closed 2023-02-10 by DarylDohner